TheGuyDanish/CM4_MATX #2: Expose both a USB pin header and a USB-A socket internally

mo-g, 2020-11-14:
This is something cribbed straight out of Supermicro's playbook, but I think we should absolutely borrow it.

The design as current exposes two USB 2.0 headers. I propose we only expose one header and expose the other as a USB type A socket.

The reason is, there are many USB devices that would make sense to use internally in a Pi based PC which are only available with A connectors and not header blocks:

1. USB-SATA adapters. Many people use these, as they connect a standard 2.5" ssd to a USB socket. With an internal USB port, one could be mounted in a normal drive cage and still used as a Pi's boot drive - handy, since NVMe boot support is not available.

2. USB flash drive. See above, except cheaper. A USB flash drive could hold the pi's bootloader only, for booting to NVMe. Allows easy use of NVMe with the Pi CM4 lite, potentially allowing the eMMC interface to be used for something more useful. (We're exposing that as a header block too, right?)

3. USB-Floppy disk drive interface. Something like this - https://www.ebay.co.uk/itm/264239344659 - Which allows an internal 3.5" drive to be mounted for retrogaming fun. But, it needs a USB-A, not a header.

As an alternative to this proposal, I'd also accept headers-only, as long as they were two physically separate four-pin headers, and not the common two-port 9-pin header block. That would allow something like a single port version of this to be made to allow both internal or external type connectors to be used in any combination: https://www.ebay.co.uk/itm/383805792699

mo-g, 2020-11-14:
Another alternative - use USB2517 instead of 2514 - two external ports, two complete 9 pin headers AND an internal USB-A.

TheGuyDanish, 2020-11-14:
There's definitely a potential for this and I was considering it when I was writing the design goals in the README. Most internal USB headers will expect the two-row header that normally is used for exposing USB 2 ports.

Using the 2517 could be an interesting fix, seeing as it should be a more or less drop-in replacement. I'll look into that.

TheGuyDanish, 2020-11-15:
I didn't mention this when I committed it yesterday, but https://github.com/TheGuyDanish/CM4_MATX/commit/08fd37ccbf002c83fe7bd2444757735594c0ec7e put a lot of my work on this into the repo. The USB2517 is placed, as are two 2.54mm expansion headers, the dual-stacked rear-panel USB connectors, and finally the vertical USB A header on the board itself. Totaling to all 7 ports. The file https://github.com/TheGuyDanish/CM4_MATX/blob/master/CM4_MATX_PCB.pdf shows where I expect to put these on the board, but would love to hear other thoughts.

There is still some work to go for USB, though. Namely:
* The VBUS rail doesn't exist yet. On the CM4IO board, the VBUS rail is made by putting 5V through an AP22653W6 power switch IC and using the output (with one decoupling capacitor per port) as VBUS.
* Power still needs to be attached to the USB2517 chip. I haven't done an exact count as of yet, but it's likely we'll need a few more decoupling capacitors for 3.3v rail on the power inputs on the USB2517 (IC1) and the FSUSB42MUX (U1).

I plan to look into this on Tuesday.

TheGuyDanish, 2020-11-24:
I believe this has large been completed. Anything that might remain now are overall steps, not anything specific to this issue, closing.

closed 2020-11-24 by TheGuyDanish