TheGuyDanish/CM4_MATX #3: Integrate an IEEE 1284 (Centronics) parallel port (not through a USB adapter)

volkertb, 2020-11-14:
Although it's considered a legacy interface, I'd very much like to see a parallel port on the backplate of this motherboard, for various (retro) tinkering purposes. One that supports bit-banging, a low-level feature that USB to parallel port adapters don't offer.

Perhaps you could implement this fairly easily by integrating @garlick's pi-parport HAT design at https://github.com/garlick/pi-parport into your motherboard design?

I'm not sure if this would take up any pins that are already required to offer other types of integrated I/O that you plan to implement, but if not, this would be really nice to have.

mo-g, 2020-11-14:
This would use up a bunch of GPIO (actually, nearly all of it) - BUT, this is exactly the kind of thing that could be done via the 40-pin connector if we mount it next to a backplane slot. You could expose a 40-pin IDC externally, my "geekport" or use this schematic plus a MAX232 to give a parallel port and a serial port as an expansion card.

This could also be used, if you wanted to keep the 40-pin connector for something else, as well:
https://www.dell.com/en-uk/work/shop/dell-parallel-serial-port-pcie-card-full-height-for-mt/apd/540-bcgv/networking

mo-g, 2020-11-14:
As a side note, I want one of these GPIO to Parallel+Serial Port cards if this board ever gets built! I can think of a lot of uses for that! :D

volkertb, 2020-11-14:
Interesting alternative suggestion. I wouldn't want to sacrifice a precious PCIe slot for it, though. Indeed, having some kind of expansion card that plugs in to the GPIO pins and exposes such ports on the backplane would indeed be the way to go.

By the way, I was about to open a separate topic for requesting an integrated RS-232 DE-9 port as well, but I guess that's part of the same conversation. (Except that an RS-232 port wouldn't require a lot of GPIO pins and therefore could perhaps be integrated after all. What do you think?)

TheGuyDanish, 2020-11-14:
There wouldn't be any sacrifice. As mentioned in #1, the switch chip only supports three downstream PCI-e lanes. Seeing as the mATX spec covers four, we can put an IDC header with the GPIO where the fourth slot _would have_ been. This would allow anyone to potentially install HAT-based expansions into that fourth slot. It would require a custom bracket/cutout to expose it externally. But there wouldn't be any sacrifice involved.

volkertb, 2020-11-14:
Yeah, you both have convinced me that that's the way to go. In that case, I guess I can close this thread, right? Since exposing the GPIO pins through an IDC header would be more flexible than hard-wiring an integrated parallel port to some of the GPIO pins.

TheGuyDanish, 2020-11-14:
I think it's safe to close this, yeah. While the idea behind this is solid. It does definitely lean more towards being one part of a greater "expansion card"/PiBus project.

mo-g, 2020-11-14:
Side note - my original thought process for this (@TheGuyDanish) might have a different perspective) is to try and expose as many of the features as possible. The reason I want us to use the i2c and i2s for the HDMI in, is because doing so allows us to make use of the otherwise useless CSI ports.

Making a DB9 serial port would remove the UART fro being able to be used internally, but like the Parallel port - you need little more than a single IC (MAX232 TTL-RS232 converter) which is a long since common chip for DIY projects. :) You could easily fit them both on an expansion card - and even have a flying lead out to a DB15 game port on the card, too. For that full 1993 experience!

TheGuyDanish, 2020-11-15:
Closing this for now as it's pretty much agreed that this is out of scope for the board itself.

(And also so we can close our first issue. 🎉)

closed 2020-11-15 by TheGuyDanish