TheGuyDanish/CM4_MATX #4: Expose SDIO as a header next to the I/O shield, not as a soldered SD Card slot

mo-g, 2020-11-14:
According to the launch blog for the CM4, by holding the SOC in a reset state, the SDIO interface can be used to access the eMMC on a full fat CM4 for programming. Outside of that case it can also be used to add an eMMC module to a CM4 Lite, as an SD I/O interface in the classic sense for external hardware (GPS, modems, etc), or.. as an SD card socket.

The CM4 Carrier hard limits all of that to JUST being used as a storage medium by using a uSD socket. We should expose it as pins, and allow the end user to attach an eMMC module, a full size OR micro SD slot via a simple passive shim, or use the SDIO functionality for another device - OR, connect it to another computer to access the eMMC.

TheGuyDanish, 2020-11-22:
I do think this is a good idea, but part of me is wondering how we could do this in an elegant way. I think for Rev.1, having a soldered SD Card slot is acceptable, mostly so we can also accommodate the CM4Lite boards that don't have on-board EMMC, but I would be willing to look into how we could work something like this out in a future revision.

Looking up examples like the ones you mentioned, GPS, modems, I see they use a full-size SD form factor. Would soldering on a full-size SD card socket make this more achievable? Given how almost every microSD card sold these days still comes with an adapter, I think it would be a reasonable option.

In a future revision (or if we have some time to implement it) something like a Fairchild/ONSemi [FSSD06-D](https://www.onsemi.com/pub/Collateral/FSSD06-D.pdf) could be an interesting option to allow either a microSD slot, or a second slot, perhaps via exposed pin headers to give the end user control.

mo-g, 2020-11-22:
Going for a full size SD card slot makes sense *anyway*. Given the issues SD cards have as long term storage devices - well, uSD and SD have largely the same capacities, speeds and price points... but uSD has a dramatically shorter lifespans. Full size SD is a much more sensible approach than micro.

I was basically picturing a header block, and a passive adapter that just wires the pins to a SD slot (probably with a right angle connector, so it sits horizontally when in a vertical case). But I think a full size SD is a decent compromise. If someone wants a bigger flash storage - NVMe is a better bet than a custom eMMC board anyway.

mo-g, 2020-11-22:
For SDIO devices - there are two types; the old PDA accessories, which were full size SD cards, and embedded devices like the Wifi interface on the CM4 itself. AFAIK these are electrically identical.

TheGuyDanish, 2020-11-22:
Just looked up a SDIO WiFi adapter and yeah, it does seem pretty similar.

In any case, I'll get on finding a decent full-size SD card holder to solder onto the board and put it on tomorrow.

mo-g, 2020-11-22:
I look forward to my first issue being closed. ;)

TheGuyDanish, 2020-11-24:
Full size SD card slot added in 9a986fd, resolving this.

closed 2020-11-24 by TheGuyDanish