TheGuyDanish/CM4_MATX #10: Can I take your licence and registration, please?

mo-g, 2020-11-15:
I'm kind of in favour of copyleft, since nice to get bug fixes from any company who makes money off this - but permissive is more likely to encourage somebody to actually make it.

Copyright assignment, or collaborative licence model? Contributors.md?

Or do you just want to CC0 (Public Domain) this?

But you should probably add a licence, anyway. :)

TheGuyDanish, 2020-11-15:
I think this should be up for some discussion. In software, I'm very much a fan of the MIT license and think it would suit this project well. (Make the changes you please, use it commercially, privately, whatever, just make sure you leave a footnote to us). I'm curious to hear what others think would be a good license to put it under.

As an aside, I'm also completely up for a CONTRIBUTORS.md, which I'm adding to my TODO list.

mo-g, 2020-11-16:
I also default to the MIT licence most of the time, or LGPL if I think it's something significantly worthwhile and complex that justifies forcing return contribution. This... while it represents a lot of work on your part, nothing in it is going to be spectacularly difficult to replicate.

I think we'd need to tweak the MIT licence as well - for one, every other word is software - we need to replace that with "hardware design, and hardware derived from it".

Also, as unlike software _patents_ are an issue for hardware designs, we should look to have some kind of "if you use, adapt or contribute to this design, you can't sue us for any infringement it makes on your patents" in the manner of MiniDisplayPort. Otherwise, we're asking for someone to make a contribution, patent it, then sue us for infringement.

We also need to be a *little* careful about borrowing designs from the IO Board Schematics. So far, both cases we've wanted to do that we've since done something completely different (USB, fan controller) but - you're probably best looking at the original IC datasheet for a reference design of the RTC IC rather than the I/O board datasheet, just in case the RPF decide we've infringed _their_ IP.

TheGuyDanish, 2020-11-16:
Regarding licensing, there is a specific "Open Source Hardware License". https://www.oshwa.org/definition/ It's a bit too late for me to read license text, but it could be a potential option.

And I do agree on the on the point of IP and _borrowing_. Reference designs will always be better than lifting it from other projects. In this instance, it is mostly done with the idea of rapid prototyping in mind, but there's definitely a concern there.

volkertb, 2020-11-16:
> Also, as unlike software _patents_ are an issue for hardware designs, we should look to have some kind of "if you use, adapt or contribute to this design, you can't sue us for any infringement it [...]

That's exactly why the [Apache 2 license](https://www.apache.org/licenses/LICENSE-2.0) is my favorite open source license: it has a patent retaliation clause. Other than that, it provides a high level of freedom in what others are allowed to do with the sources. Of course I'm no legal expert, and I'm not sure how well a software-specific license such as the Apache 2 license would apply to hardware designs.

Perhaps there is an open source hardware license out there that also has a patent retaliation clause?

garlick, 2020-11-16:
I've used CERN OHL 1.2 for some of my hardware projects FWIW, and it says something about patents:

https://ohwr.org/project/cernohl/wikis/FAQ#q-i-find-the-section-on-patents-a-bit-confusing-can-you-explain-to-me-in-plain-words-what-you-are-trying-to-achieve

(since this is my first time speaking up here, I also wanted to say "cool project!" :-)

funkyguy4000, 2020-12-09:
I would love to contribute to this project although I think the best avenue here is a permissive license like MIT or BSD.

TheGuyDanish, 2020-12-09:
Cheers, Shannon. Always happy to have more people on board! (pun not intended)

I'll be going over this on the weekend again and comparing options. So far, I'm leaning towards the CERN OHL, with a weakly reciprocal model (OHL-W). If my understanding of the OHL is correct, weakly reciprocal would mean that any changes made to the base design, schematics, etc, would need to be published. But any other additions (say new ICs on the board, for example for SATA or something along those lines) would not. I feel that's a fair distinction.

We could also apply OHL-P, which, hugely simplifying all the license texts, would be the BSD-equivalent of the hardware world. Anyone can take it, relicense, and use it without needing to distribute sources. In reality, I'm also fine with this option, but would like to see improvements directed back to the original repo.

The only hard stance I have at the moment is that I won't choose OHL-S, as I'm not really big into the idea of forcing anyone using parts of this in their own projects to publish _all_ of their design.

The relevant summary can be found in [the OHL wiki.](https://ohwr.org/project/cernohl/wikis/faq#q-what-are-all-these-suffixes)

funkyguy4000, 2020-12-12:
Sounds good! I wasn't aware those licenses exist. Personally I'm more of a permissive kind of guy.
(Also since I didn't introduce myself last time, hello! Really like this idea)

TheGuyDanish, 2020-12-14:
As of 371aa42, this project is now licensed under CERN OHL-P v2 allowing permissive use of all designs.

closed 2020-12-14 by TheGuyDanish