TheGuyDanish/CM4_MATX #5: Adding PoE Support

derwana, 2020-11-15:
YES i am perfectly aware of the ATX formfactor and the ATX-PSU coming with it. But i was thinking about all the other possibilities of powering a board: onboard MicroPSU or even only PoE.

I am currently diving into `IEEE 802.3bt` on PoE++ / 4PPoE which could give up to 70 effective Watt.

Nevertheless: i would really appreciate compatibility for a PoE HAT for not so power gussling usecases.

volkertb, 2020-11-15:
I'm not sure this project would fit your use case. The MicroATX form factor is intended to provide a desktop computing solution, which would be built in an ATX or MicroATX case, and include a PSU.

The point of PoE is that a compact device such as a WiFi router or perhaps some IoT device could be installed on a wall or ceiling somewhere, with a single wire providing both power and connectivity. I don't see the point in enabling something like that for a desktop PC case.

I do agree with you, however, that it would be very useful to have PoE support integrated in the regular Raspberry Pis, eliminating the necessity for a PoE HAT, and also offering a more reliable alternative method of powering a Pi than through 5V USB AC adapters, many of which are of questionable quality and often don't provide enough power for continuous stable operation.

mo-g, 2020-11-15:
Hmm..

I've actually thought about this as well. Most PoE modules are designed to output 12V. 12V at 35W, plus 5V at 15W would power this board, the SOM and a single 1 lane PCI-e card. So it's not unreasonable.  Something like an AG5800 module could potentially even drive a GPU and the SOM together. But, I think the additional cost and board space of building support for such a module is likely to be high for something that is specifically meant as an ATX board.

If you're using PoE for power, you're not realistically going to be using more than one PCIe card anyway - which means that a board like this would be wasted. But I'd be interested to see a poe+ enabled ITX "Routerboard" based on this but without the PCIe switch and a single x16 slot positioned for use with a passive single-slot riser in a 1U case. Would be neat as a drop-in for something like a Supermicro appliance.

TheGuyDanish, 2020-11-15:
I think this is a good option to keep in mind, but I don't think it's a "Revision 1" feature. I'm mostly concerned about people trying to use PoE to power the board, then getting upset when it randomly shuts off because there's not enough power. Assuming the user is educated enough about the limits, it could be an interesting prospect.

derwana, 2020-11-16:
> I think this is a good option to keep in mind, but I don't think it's a "Revision 1" feature.

@TheGuyDanish i am totally with you. Please add a handy label to this issue.

TheGuyDanish, 2020-12-28:
Closing this in favor of #47.

closed 2020-12-28 by TheGuyDanish