TheGuyDanish/CM4_MATX #6: Adding Fan Control Circuitry

derwana, 2020-11-15:
Seems overkill on first glance – but the CM4 might get very toasty on heavy loads. Maybe place a thermistor somewhere near the maximum heat spot under the CM4 to control that circuitry.

TheGuyDanish, 2020-11-15:
Fan control is definitely a good idea. Given that we have +12v from the ATX power supply, putting some [4-pin MOLEX fan headers](https://www.molex.com/molex/products/part-detail/pcb_headers/0470531000) on the board would be easy. If we can find a way to do PWM control from the Pi itself, we could even let the fan speed ramp according to CPU temp. Given that there's only two PWM pins on the Pi, I reckon something like the [MAX31790](https://datasheets.maximintegrated.com/en/ds/MAX31790.pdf) would be a reasonable addition. Six channels should definitely be enough to control all the fans a user would really need. But it does drive up cost.

derwana, 2020-11-15:
I am split in two. On the one hand i like the idea of reading the temperature and controlling the Fans via PWM for obvious reasons. On the other hand i am a simple guy: sticking a thermistor to the SOC and reading it with a [MAX6615 Dual-Channel Temperature Monitor and Fan-Speed Controller](https://www.maximintegrated.com/en/products/sensors/MAX6615.html). No PWM and additional programming required. Also it is [2.31$](https://www.maximintegrated.com/en/products/sensors/MAX6615.html/tb_tab3) (without the thermistor) compared to [2.70$](https://www.maximintegrated.com/en/products/interface/sensor-interface/MAX31790.html/tb_tab3) for the MAX31790.

TheGuyDanish, 2020-11-15:
I think the simplicity of the MAX6615 would be favored in this scenario. While the CM4 does get hot, I don't see an instance in which having 6 fans instead of 2 would make a remarkable difference. Setting a front and rear fan in one's case and a heatsink on the CM4 components should be good enough.

Do you want to have a stab at adding it to the schematic, or should I put it on my to-do?

derwana, 2020-11-15:
I am going to read the whole datasheet and will do a circuit for it – but may take some days. SMD or THM?

TheGuyDanish, 2020-11-15:
Not in particular. At this point in time, whatever you think is appropriate and decently priced is good. My plans for the (eventual) manufacturing of this board would be taking the board design and BOM to a vendor like PCBWay and getting quotes for getting the board printed and assembled. Either with us supplying parts, or them sourcing them. Which is also why I'm eager to keep costs down.

TheGuyDanish, 2020-11-15:
I'm hoping to take a break today and tomorrow from changing the schematics, so you've got a bit of time to run on. 😂

geerlingguy, 2020-11-15:
Note that the Pi IO Board itself just has an I2C PWM chip in line with the 4-pin fan connector. I haven't found a way to control it yet (mostly because I'm such a noob with I2C and there aren't any libraries for the chip they're using AFAICT). But that would be an option too: [CM4 fan controller lm-sensor](https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=290795).

TheGuyDanish, 2020-11-15:
Good point, Jeff. The one built-in on the CM4IO is an SMSC EMC2301-1-ACZL-TR. The only downside of implementing it would be that it only supports a single fan. From what I can see, 2302, 2303, and 2305 exist as well, each supporting more fans (what the last model number symbolizes). All at decent prices. If we do decide to go down the i2c fan controller route later, an EMC2305 ([~€1.58](https://eu.mouser.com/ProductDetail/Microchip-Technology/EMC2305-1-AP-TR?qs=%2Fha2pyFadui5v9hCMAZpa%2Frt92tib3bYXatmezcE77%252Bls%2FnqSHTE%2FQ%3D%3D) could be a good option.

TheGuyDanish, 2020-11-15:
Forgot to attach a screenshot, oops.
[image]

mo-g, 2020-11-15:
Stupid question - why would we need more than one fan?

A single fan will cool the SOM. Most uses of our HDMI-CSI and PCIe switch chips use no cooling, and an ATX case is going to have plenty of air for passive convection for those - plus anyone using SATA, SCSI, SAS or a GPU is going to be using a full PSU with a built in fan anyway, and overclockers.co.uk lists 36 different USB fan controllers that can be hooked up with our three internal USB ports. Given that the TDP of the BCM2711 is 7.5W, I don't see us making enough heat to need more even if this was in some kind of 1U case.

If the cost is negligible, a 2303 could make sense for front and rear case fans as well, but this seems pretty overkill. ;)

Says the person who worked out how to add a DisplayPort interface for triple monitors...

TheGuyDanish, 2020-11-15:
I do see your point, but consider how toasty the Pi4 got around release. 7.5W can still produce enough heat to thermal throttle it. I'm not sure if @geerlingguy touched on the temperature of the CM4 in any of his videos, but I imagine it _could_ be a potential throttling situation. In truth, we probably won't need fans. If a user wants to cool their CM4, chances are a passive heatsink would be enough to take care of the worst throttling. But at the same time, it's a small feature to add, and relatively inexpensive at that.

Unlike two Toshiba HDMI-CSI2 bridges. ;)

mo-g, 2020-11-15:
Point taken! Mouser said about 30¢ extra for the bigger IC, and we're not talking much more than that for the 4 pin connectors, so I'm in favour, I just don't know if we can actually *justify* a second or third fan connector on the basis of actual thermal needs. One is probably plenty - but YOLO.

If we can add an RGBI header, I'm also in favour of that.

geerlingguy, 2020-11-16:
Note that at the default clock, with natural convection and _no_ case, I did not see the Pi throttle (it stayed under 70°C the entire time I was testing it).

Overclocked to 2+ GHz, though, it would overheat after 30+ minutes of `stress-ng` if I didn't have some sort of air flowing over the chip. Adding a good heat sink would probably be adequate for passive cooling (assuming the case allows enough airflow).

Note that many PSUs have built-in fans to cool themselves, and most cards like GPUs that require more power and generate more heat have their own built-in fans too.

So one fan, IMO, would be plenty—the only use case where I see more being nice is if someone is putting in a ton of beefy hot hard drives for a NAS, and they want one fan for the drives and another for the Pi.

derwana, 2020-11-16:
> So one fan, IMO, would be plenty—the only use case where I see more being nice is if someone is putting in a ton of beefy hot hard drives for a NAS, and they want one fan for the drives and another for the Pi.

I'd consider myself as one of those NAS guys, building one with leftover HDDs and SSDs. Currently i got one in a 1U case and i would be happy to have more than one fan header. Only one connector for the fans in the front and back is quite messy with all the splitters and extenders.

mo-g, 2020-11-16:
Didn't consider drive cooling, I'm sold.

derwana, 2020-11-21:
Over the last days i did some research on fan control ICs and came to the conclusion that [EMC2301](https://www.microchip.com/wwwproducts/en/EMC2301), [EMC2302](https://www.microchip.com/wwwproducts/en/EMC2302), [EMC2303](https://www.microchip.com/wwwproducts/en/EMC2303), [EMC2305](https://www.microchip.com/wwwproducts/en/EMC2305) got the best price performance ratio so far (the [EMC2301](https://www.microchip.com/wwwproducts/en/EMC2301) is also on the Pi IO board). They can control 1/2/3/5 fans with different PWMs and are addressable over I²C compatible SMBus. But as Jeff mentioned:
> [..] there aren't any libraries for the chip they're using AFAICT. But that would be an option too: [CM4 fan controller lm-sensor](https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=290795)

On the other hand we got chips from Maxim like the [MAX31740](https://www.maximintegrated.com/en/products/interface/sensor-interface/MAX31740.html) or [MAX6615](https://www.maximintegrated.com/en/products/sensors/MAX6615.html) who use NTC thermistor inputs to adjust the PWM fan signal. The first on is completely based on a voltage divider with the thermistor, does not require any configuration over I²C or similar and got one PWM Out. The second is more feature rich with fan-fail-safe protection, programmable fan-control characteristics and two PWM Out.

When the project is fully rebased to a stable KiCad version i'll do a circuit for MAX31740, MAX6615 and EMC2302 each – so we get some variety to choose from.

TheGuyDanish, 2020-11-21:
I've been making a lot of changes and think we're about to be at a point where the rebased branch is more developed than the Nightly one. So expect rebase to be merged into master at some point tonight or tomorrow.

derwana, 2020-11-22:
the three circuits are done – do you want them as separate pull request or as one?

TheGuyDanish, 2020-11-22:
Separately would be nice, if it isn't too much of a hassle. We can keep discussion of them to this issue.

TheGuyDanish, 2020-11-22:
Looking over the different chips and their datasheets, I'll admit I'm more drawn to the MAX6615, with the EMC2302 a close second. Looking at pros and cons, I think the 6615 would be nice from a "set it and forget it" perspective. We don't _need_ to put it on the SMBus and we can let it act off of the thermistors. But that is also quite inflexible, whereas with the EMC2302, there would technically be better options towards finer control.

The EMC2302 does concern me with the lack of seemingly any mainline kernel driver support. It appears a patch was submitted, but denied, looking at the LKML.

I think the best of the bunch among these three would be the 6615, but at the same time, I'd be enthused to see a fully-featured controller like the 2302 on the board, but with mainline drivers and support in lm-sensors. Perhaps the [MAX6639](https://pdfserv.maximintegrated.com/en/ds/MAX6639.pdf).

Any thoughts on that?

(I managed to find a list of lm-sensors supported hardware monitoring chips [here](https://www.kernel.org/doc/html/latest/hwmon/index.html#hardware-monitoring-kernel-drivers) in case we want to do more digging on this. [See for example the MAX6639 page.](https://www.kernel.org/doc/html/latest/hwmon/max6639.html))

derwana, 2020-11-23:
Call me stupid - i missed out that list completely, although it is in my bookmarks. I'll revisit this later today.

TheGuyDanish, 2020-12-22:
@derwana Have you had any chance to look at this? I'm starting to look into the MAX6639 as a potential candidate but I'm trying to get a hold of one to do some testing with (Which might end up being in the new year). Just curious.

derwana, 2020-12-22:
Yes i have. One with MAX6651 is finished, but i am not satisfied with it at all. The MAX6639 and maybe MAX31790 are the next ones on my list.

TheGuyDanish, 2020-12-22:
Okay, cool. I think the 6639 is a good one to go for, considering the support it has and the seemingly simple layout. In that case, I won't look too much further into it right now. Cheers mate.

TheGuyDanish, 2020-12-23:
Closed by #36

closed 2020-12-23 by TheGuyDanish