TheGuyDanish/CM4_MATX #8: Consider a RTC solution that interfaces with USB, to save GPIO pins

volkertb, 2020-11-15:
Wouldn't it be better to integrate a USB-based RTC into the motherboard design, as opposed to a GPIO-based one, to save pins? 

Something like this perhaps: https://easyeda.com/rodyne/USB_RTC-dfce249ec7af4d7380d69c51a7e2cfec

USB (at least theoretically) supports up to 127 concurrently connected devices. Therefore, whenever integrating any functionality on the motherboard that can be implemented over either USB or GPIO, USB should be the preferred way to go, right?

TheGuyDanish, 2020-11-15:
While I think the theory behind is good, some things are just not worthwhile to integrate over USB. While I'm sure USB-RTC is something that works, I think acquiring it would be more expensive than implementing the CM4IO way of doing RTC via a dedicated chip and battery.

The PCF85063AT IC that is used on the CM4IO board to provide RTC is $0.69 for a single chip. That, combined with the fact that it doesn't use up any GPIO pins (since i2c is a bus), makes it a poor candidate for replacement. I doubt one of these USB-RTC devices could beat $1-2 in price.

As it stands right now, my current plan is to wait for #6 to be completed, then get the PI7C PCI switch integrated properly. With 3 exposed PCI-e slots, the GPIO header, and 7 USB ports, we should have more than enough IO for any early-adopter of the CM4 to toy around with.

Once the items mentioned above are integrated (and I've hooked the USB hub up with power), we can place the PCI-e slots and GPIO header to get traced on the PCB. After that, we could get a first quote on what board assembly would run up to.

For any board assembly quote, I plan to open an issue and tag all active participants, highlighting the features, slots, pricing per board and the total price for the smallest feasible batch. We can then discuss if we'd like to make an order for that board revision or wait until we've had more time to work on it.

volkertb, 2020-11-15:
Ah, it works over i2c and  doesn't take up any GPIO pins? In that case, forget I mentioned this. Thanks. :sweat_smile:

closed 2020-11-15 by volkertb