TheGuyDanish/CM4_MATX #9: Perhaps use open-ended PCIe x1 slots instead of PCIe x16 slots?

volkertb, 2020-11-15:
Wouldn't it be preferable (and equally flexible) to use open-ended PCIe x1 slots instead of full-blown PCIe x16 slots that would be wired up as x1 slots? 

[image]

It would make it clear that these are indeed just x1 interfaces, and there would also be a lot less solder points required, so perhaps it would simplify the board design a bit as well?

The one challenge here might be that such slots could be difficult to source, at least judging from [this Reddit discussion](https://www.reddit.com/r/pcmods/comments/5xav00/purchasing_an_openended_pcie_connector/), although it's 3 years old, and availability may have improved since then.

TheGuyDanish, 2020-11-15:
Hmm. I'll look to see if I can find them anywhere. Until then, can someone figure out what Red Shirt Jeff's hourly rate is? 😂

volkertb, 2020-11-15:
I'm sorry, I didn't get that reference at all. :sweat_smile:

mo-g, 2020-11-15:
@volkertb It's a Jeff Geerling reference.

TheGuyDanish, 2020-11-15:
Red Shirt Jeff is Jeff Geerling's altar ego, to whom the Dremel is the answer to all problems.

volkertb, 2020-11-15:
Ah, okay. I didn't start following him on YouTube until recently. But a search for "Red Shirt Jeff" on Google didn't turn up his YouTube channel yet.

By the way, I tried to cut an opening on a regular close-ended PCIe x1 slot a while back. My hacky method was to heat up a small screwdriver with a soldering iron and then melt open a slit. I didn't have a dremel on hand. Unfortunately, I didn't do a good job at it, since it ended up not working. The right tool for the right job, I guess. :grimacing:

Honestly, I don't understand why not all motherboard manufacturers just use the open-ended variants of the x1 slots these days.

mo-g, 2020-11-15:
Two reasons why not:

First, it offers more physical support. So, if you've got a big heavy GPU in a tower case without a support at the far end of the card, a 1x slot means a lot of leverage on the slot and backplane. A 16x slot distributes the load better.

Second - a 1x slot and a 16x (1x electrical) slot are not electrically identical. While the 16x slot still only has the same signal conductors, it gets additional power and ground pins. Officially, a 1x slot can only offer 25W whereas a 16x (1x electrical) slot can offer 75w. This is why 1x to 16x risers have a Molex on them - to provide the additional power rails. Since we're expecting people to plug more than a 1x USB 3 card in here, we should really go for the full wattage slots. Otherwise people couldn't put (for example) a GT 1030 card in here - the open 1x slot wouldn't provide enough power and there's no-where to put a powered riser card in the ATX case.

TheGuyDanish, 2020-11-15:
Having looked through Mouser, Digi-Key, LCSC, as well as a few manufacturer pages. Open-ended x1 ports seem near-impossible to find. :/ That, along with the points @mo-g makes illustrates pretty well why x16 would still be a better choice, compatibility-wise.

geerlingguy, 2020-11-16:
I think the major thing I would want in either case is good labeling / silkscreening on the board—the IO board has a very nice "PCIe x1" label next to the slot that makes it immediately obvious it's a 1x port (besides the physical size).

[image]

I agree with @mo-g that a 16x slot is nice for the added support. Red Shirt Jeff has been doing some testing with different slots and found that it's annoyingly unstable to plug in heavier cards with big heat sinks/fans built into shorter slots, because they have a tendency to lean away from the slot as @mo-g mentioned.

mo-g, 2020-11-16:
Sensible. I'd *like* to think that anyone who knows enough to be looking at this would know they're 1x slots without a label. But unfortunately, I'm optimistic - not naive.

volkertb, 2020-11-16:
You've all made some good arguments against this. PCIe x16 slots it is! Closing this issue.

closed 2020-11-16 by volkertb