TheGuyDanish/CM4_MATX #11: More PCIe Lanes

mo-g, 2020-11-16:
@TheGuyDanish I know this is a wontfix for you from our original discussion on Geerling's 19, but I thought it's the kind of thing that should be an issue *anyway* - and maybe someone decides to raise a PR.

We should swap from PI7C9X2G404SL to PI7C9X2G606PR to give us two extra PCIe downstream ports - this would allow us to lose the MUX chip and also frees up a spare lane for USB3, Thunderbolt, a fourth PCIe slot or a second m.2 slot for a TPU or other minicard.

TheGuyDanish, 2020-11-16:
As ever, I'm happy if someone wants to look into drawing the schematics for this. If it gets done, cool. If not, I'm still planning on going forward with a "rev 1" board with the 404SL. When we have a more "full" board that works, I could potentially look into adding the 606PR myself. At the moment, there's just a lot of other stuff I'd rather focus my attention towards. I'll label this as 'help wanted' for now.

mo-g, 2020-11-16:
By the way, I hope you don't mind all my "concept" PR's.

I'm crap with answers, but I'm good at thought-provoking questions. ;)

mo-g, 2020-11-16:
> I'm happy if... "rev 1" board with the 404SL ... I could potentially look into ... rather focus my attention

100% how I was thinking this. Just thought it should be documented somewhere. :+1:

TheGuyDanish, 2020-12-28:
Closing in favor of #48

closed 2020-12-28 by TheGuyDanish