TheGuyDanish/CM4_MATX #14: Export KiCad design to Blender to generate a high-quality near-realistic render of the end product

volkertb, 2020-11-16:
As an improvement on https://github.com/TheGuyDanish/CM4_MATX/issues/12, perhaps we can export the KiCad designs to Blender and render a near-realistic image of what the board would be expected to ultimately look like.

There appear to be multiple projects on GitHub that aim to facilitate just that, including one that was featured on Hackaday here: https://hackaday.io/project/175077-remoticon-kicad-to-blender-pcb-renders

The results look pretty impressive.

If we could manage this for this project as well, it would attract a lot of interest and publicity, including (hopefully) more knowledgeable volunteers willing to help us with the hardware design.

I'll look into those projects to see how feasible this would be. Seems like a fun challenge, and something that even I, lacking hardware development skills, can take a stab at. :slightly_smiling_face:

TheGuyDanish, 2020-11-16:
If anyone wants to do this, I'd love to see it. Ultimately, I don't think this is something that I'd put a whole lot of effort into, seeing as the KiCad renders will likely be fine for most things. But as ever, contributions are welcome!

derwana, 2020-11-21:
this link might also be helpfull: [https://wyolum.com/experimenting-pcb-rendering-using-kicad-freecad-and-blender/](https://wyolum.com/experimenting-pcb-rendering-using-kicad-freecad-and-blender/)

volkertb, 2020-11-23:
@derwana Thanks. I based my rendering on the instructions at https://github.com/Hack-a-Day/Remoticon_KiCad_Blender, which also featured at [Hackaday](https://hackaday.io/project/175077-remoticon-kicad-to-blender-pcb-renders).

Now the examples you see there are of much higher quality than what I managed to whip up, but I'm not that skilled with 3D modeling. I've dabbled a little bit with Blender, but I'm by no means a wizard. :sweat_smile:

@TheGuyDanish I also included the Blender file, since importing the PCB into Blender proved to be a bit more challenging that I had expected. First of all, color gets lost when you import VRML files (exported from KiCad) in newer Blender versions for some reason. Importing it in Blender 2.79 preserves the colors, but the VRML exported from this project is so large that the Linux version chokes on it and crashes, so I had to import it in the Windows version of Blender 2.79. Once the export was successful, I saved it into a .blend file, which I could them open in Blender 2.90 without problems, and with the colors still there.

With some more time (and or contribution from an experienced 3D modelling artist), I'm sure an even more impressive 3D rendering can be made of the PCB design, like the ones you see on Hackaday. However, the .blend file (even being compressed) is quite large at 13MB. I therefore completely understand if you prefer not to have the .blend file included in the code base. If so, let me know, and I'll update the Pull Request accordingly.

Thanks. :slightly_smiling_face:

derwana, 2020-11-23:
@volkertb from the link i posted (regarding the colorloss with Higher blender versions):
> A workaround to this that I stumbled upon is to use the online 3D file format converter at CADexchanger and convert the KiCad VRML file to Collada DAE format. Importing the DAE format in Blender results in glorious color renders.

Unfortunately the converter is capped on 10 conversions per month on a free account.

derwana, 2020-12-23:
HACKADAY just posted an online workshop titled _KiCad to Blender: Photorealistic PCB Renders_ on [Youtube](https://www.youtube.com/watch?v=mVpT5dw4sew). It's worth a watch i think.

volkertb, 2020-12-26:
At first I tried automating the process of importing the VRML file in Blender and rendering it, by scripting it with Python. But I couldn't get it to work. The embedded Python interpreter in Blender seems finicky and the Python API is poorly documented and doesn't appear to expose all of Blender's functionality.

That's a shame, since having a script for this would make it so much easier to redo the render any time the KiCad design were updated.

Help with this would definitely be welcome!

derwana, 2020-12-26:
I'll have a look into it the next days.

Bentschi, 2020-12-27:
Hi,

I tried what was explained here with my PSOne RPi Compute Module board.
This is my result... (the board is not done yet)

[image]

If you guys like it and need help with that let me know, i could do a render or 2 for you.

ldrrp, 2020-12-27:
It would be cool to use github actions and script the rendering? So its always up to date

Bentschi, 2020-12-27:
> It would be cool to use github actions and script the rendering? So its always up to date. Although not certain the amount of compute that would take.
Scripting the actions on github is beyond the scope of my knowledge, but it took me less than 2 hours to export from kicad and set up the scene in blender and making the model that shiny. And with the totally cranked up settings i use to render it takes 18.05 seconds and 476.18 MB ram on the CPU with Eevee (I am not a fan of Cycles).

TheGuyDanish, 2020-12-27:
As I've mentioned before, I think a render like this is nice to have, but ultimately something we only should produce every now and then. Maybe on major revisions or release tags.

Unlike the schematic and PCB PDFs, the render can't be updated in one or two minutes by anyone who is already working on the main schematic and PCB, which is why I'm erring on putting more effort towards it. Maybe documenting the process, perhaps putting it into a documentation file, then someone can do it, as they please, when we reach a major milestone.

Bentschi, 2020-12-27:
It is already pretty well documented how you would go about exporting from KiCad and import that in Blender:
https://hackaday.io/project/175077-remoticon-kicad-to-blender-pcb-renders/log/185976-getting-from-kicad-render-to-blender
And yes, as mentioned in that article, I had to use an older version of Blender (2.79) to get the imported model right, and then used a newer version (2.91.0) to actually render it.
I could provide the Blender scene with all the settings I used to render and without the model, if interested, but it want look as good if the materials are not tweaked on the model.

volkertb, 2020-12-27:
If @Bentschi can manually make such a shiny high-quality rendering of the CM4_MATX board that would blow my current rendering out of the water and make people salivate over the end product, I'm all for it! :sweat_smile: 

Ultimately, it would be great if we could get the render process to work with a simple Python script that we could occasionally run whenever major changes or improvements are made to the design. Full automation might be problematic, since I believe you'd need the embedded Python interpreter distributed as part of Blender.

I've tried to make such a script (with the Blender Python bindings) work standalone, with no success so far. Again, I found the documentation w.r.t. the Blender Python bindings to be quite lacking.

Bentschi, 2020-12-27:
I tried, but not really sure... it looks a bit boring... maybe because there is so much room left and no traces yet.
Had to align the PCIe 3D model and had to remove the ethernet port to export it.

[image]

and here are the Blender files:
https://github.com/Bentschi/CM4_MATX-render/blob/main/CM4_MATX-render.zip

Bentschi, 2020-12-28:
I thought a little about the "boring look" of the picture.
I tried to catch the eye mostly on the upper left corner with shiny metal and the lighting, but as a whole it doesn't look "real" to me.
But it could be better if the background would actually be a atx computer case, so the eye has more to focus on the important bits.
I am not thinking of a complete case, just the important bits that are actually visible in the picture, and it would look nice i guess when a bit of lighting shines through the PCIe slots

TheGuyDanish, 2020-12-28:
I think for now, either removing the rendered image we have already, or using a screenshot from KiCad's existing 3D viewer is a better course of action. As I mentioned, it's not like our board is in a great state to be shown off anyway, so putting effort into a nice, fancy-looking 3D render of it at this point is probably not worth it, considering how many substantial changes it is still likely to go through.