TheGuyDanish/CM4_MATX #17: Rebase the project to KiCad 5.1.8

TheGuyDanish, 2020-11-17:
As was seen in #15, the nightly builds of KiCad have some issues with save compatibility between operating systems. A great way to avoid this is to use stable versions of the software. (I'm gonna take the fall for that one.) And we should make an effort to move CM4_MATX towards using a stable version.

Given we already have a bit of a base to work from, it would be nice to be able to somehow export the current work into a format that will then be readable by stable, but I doubt that will be the case. I've started a local rebuild of the project and I'm currently gathering all the schematics and footprints required to get it going.

My first challenge is that the CM4IO board, from where the actual Compute Module 4 schematic and footprint was taken, uses KiCad Nightly as well. Meaning that the resources taken from there are incompatible with stable. 

While these are issues that could be worked around, it would take some time to figure out.

I'm looking into this as a priority so we don't have people wasting time building schematics or designs that cannot be directly transferred to the project.

volkertb, 2020-11-17:
Are you sure this is worth the effort? To quote the README of the CMIO KiCAD sources:

> These files have been created in prerelease version 6 of KiCAD. If Version 6 isn't released yet then to use them you will need to download a nightly build of KiCAD. NB Nightly builds my have issues so please check the issue tracker.

That would imply that the Raspberry Pi developers purposely picked the nightlies, anticipating an upcoming KiCAD 6 release, and perhaps wanting to take advantage of the new features in that version.

Your project appears to be opening just fine with the nightly builds of KiCAD I've tried so far.

TheGuyDanish, 2020-11-17:
I'm not too sure at the moment. The current "due date" on their Gitlab for 6.0 Epics is December 31st. That's a lot of time to wait for a workable release, and I don't feel like it's reasonable to ask people to use a Linux version of KiCad because "it's the one that works with the project files". It's a workaround, but I don't think it's a viable one. Most of what's currently in the repository was put together by me over a weekend with reference to other works. I could do that again.

TheGuyDanish, 2020-11-19:
@derwana @ddkaiser @garlick (Sorry to tag you directly, gents)

After nine hours of work with very few breaks, I've made the first commit in the kicad-5.1.8-rebase branch (9a5b4a5).

Rather than trying to figure out a way to convert schematics, PCBs and symbols, I opted to do it the hard way, as I found the only symbol we really needed was the CM4 itself (annoyingly). But I have created a symbol for it and discovered a footprint made by @aep that looks very close, though I haven't made exact measurements. (His original post about it can be found here: https://github.com/raspberrypi/documentation/issues/1717). From there, I have drawn a lot of the schematic content from the old version, though more is still to go that I'll attack in the coming days. The PCB is currently back to square 1, but again, the only thing we had in there that is any kind of bugger to redo is the mounting hole placement.

In addition, this commit also includes a revised CONTRIBUTING.md in order to take us forward with #18.

My ask for you, whenever you have the time, would be to check out the branch, ideally in as many different KiCad installations as you can and ensure we have good compatibility and that I haven't missed anything when rebasing it. If it all looks good to you guys, we can merge it into master and work on that going forward.

I know some components and footprints are missing from the old CM4_MATX library used in Nightly. But I reckon some of them are better added once we actually put them down in the schematic, as well as having changed a couple of components here and there.

It's 1 AM now, so I'm gonna get some well deserved rest.

derwana, 2020-11-19:
Cheers mate! Great work!

Currently i am on version `5.1.7-a382d34a8~87~ubuntu20.04.1, release build`.

On loading the footprint editor the error  `Footprint library path "C:/SamacSys_PCB_Library/KiCad/SamacSys_Parts.pretty" does not exist` gets thrown – but can be solved by removing it from the Project Specific Libraries.

Otherwise there are no errors and missing items on my version you haven't mentioned already (some schematic content, mounting holes). I'm going to redo the latter in the next couple hours.

TheGuyDanish, 2020-11-19:
D'oh. I must have forgot that. Oh well, if you're pushing the PCB at some point, I can piggyback removing that library off of your commit. 😂

derwana, 2020-11-19:
can confirm it is working too on Ubuntu with version `5.1.6+dfsg1-1, release build`

garlick, 2020-11-19:
I'm running `kicad-5.1.8-202011012012+db9833491~87~ubuntu20.04.1` from the [kicad ppa](http://ppa.launchpad.net/kicad/kicad-5.1-releases/ubuntu).

No real problems browsing around, although, after annotating, I did get this error when I opened the "assign footprints" dialog in eeschema:

[image]

Maybe nothing to worry about but wanted to mention it.  If I agree to the conversion it looks like it just tacks on ` CM4_MATX:` to the front of project footprint refs in the `.sch` files?  Here's an example (ignore annotation related changes):
```diff
diff --git a/KiCad/CM4_USB2_HUB.sch b/KiCad/CM4_USB2_HUB.sch
index 7793ba4..f8bc519 100644
--- a/KiCad/CM4_USB2_HUB.sch
+++ b/KiCad/CM4_USB2_HUB.sch
@@ -14,12 +14,12 @@ Comment3 ""
 Comment4 ""
 $EndDescr
 $Comp
-L CM4_MATX:USB2517-JZX IC?
+L CM4_MATX:USB2517-JZX IC2
 U 1 1 602002E2
 P 1250 2600
-F 0 "IC?" H 4350 3100 50  0000 L CNN
+F 0 "IC2" H 4350 3100 50  0000 L CNN
 F 1 "USB2517-JZX" H 4350 3000 50  0000 L CNN
-F 2 "QFN50P900X900X100-65N-D" H 4300 3800 50  0001 L CNN
+F 2 "CM4_MATX:QFN50P900X900X100-65N-D" H 4300 3800 50  0001 L CNN
 F 3 "http://ww1.microchip.com/downloads/en/DeviceDoc/2517.pdf" H 4300 3700 50  0001 L CNN
 F 4 "USB 2.0 Hub Controller 7-Port QFN64 USB2517-JZX, USB Controller 7-channel 480Mbit/s USB 2.0, 3.3 V, 64-Pin, QFN" H 4300 3600 50  0001 L CNN "Description"
 F 5 "1" H 4300 3500 50  0001 L CNN "Height"
```

TheGuyDanish, 2020-11-19:
Yeah, that doesn't seem too concerning to me. I might go through them all on my next browse around the project and make sure that conversion has happened, just so no one is confused by it.

aep, 2020-11-19:
> But I have created a symbol for it and discovered a footprint made by @aep that looks very close

fwiw, i can verify it works https://twitter.com/arvidep/status/1329457736719994886/photo/1

TheGuyDanish, 2020-11-19:
@aep Thanks, and thank you as well for stopping by! It was on my todo list to shoot you an email to ask if you'd have any problems with me snatching that footprint for use in this project. Thank you for the work in recreating the footprint!

aep, 2020-11-19:
> with me snatching that footprint for use in this project

not at all. looking forward to your board. In fact i might steal back because i need ATX power supply later :D

PR #23 (2020-11-22): Merge Kicad 5.1.8 rebase into master
Fixes #17 and allows us to put more changes into gear.
commit 9a5b4a5f (2020-11-22): Where do I even start...

Rebasing the entire project to use KiCad 5.1.8 stable instead of nightly.
Most of the progress made on the schematic still in place, with a few exceptions that I can handle over the next few days.
Additonally. Resolves #17, #19, and parts of #18 (I still want to build some automated checks into the project once we are at a more advanced state)
closed 2020-11-22 by TheGuyDanish