TheGuyDanish/CM4_MATX #18: refine project workflow

garlick, 2020-11-17:
> Considered a best practice by [some](https://hackaday.com/2017/05/18/kicad-best-practises-library-management/).

I agree with Jim G here on the library best-practice.  I have been able to clone others' projects more successfully when they had the Project Library included with the project source.

When having multiple editors working in KiCad, with a shared workspace (i.e. Github), having a 'Project Library' for symbols & footprints is common.  This practice also places these parts under version control (Github) and they "live with the project" as time goes by, leading to easier recovery of old projects.

@TheGuyDanish perhaps you could make the library management conditions part of your CONTRIBUTING doc, i.e. one of these two workflows (or your own)?

Workflow Idea 1 (always use Project Library)
- when placing a symbol or referencing a footprint, only reference the CM4_MATX Project Libraries, and migrate components from standard libraries to the project within the working session.

Workflow Idea 2 (use standard libraries, and 'freeze' components to the project library on release):
- Standard symbols and footprints can come from standard libraries, but if adding a new custom part, it must be in CM4_MATX libraries.
- When moving to tag a release in Github (a major milestone on the project perhaps?) symbols and footprints are copied from the standard locations to the Project Library versions.  There may be some Python scripts to automate some of this, or perhaps the 'Edit Symbol Library Links' and 'Assign Footprints' menu options within eeschema (I've done search/replace copy/paste for these fields on a medium sized, 500 component, project.  It takes some time.)

You could (over time, and as you see fit to do so) add other best practices, for example:
- SCH files must pass ERC before sending in a pull request.  (or exception is noted in the pull request comments)
- PCB files must pass DRC before sending in a pull request.  (or exception is noted in the pull request comments)
- Files must be named as (whatever file-naming standard) i.e. CM4_\<component_name\>   (Looks as this has been done well so far)
- Subsheet files should be in the same directory as the .pro and toplevel .sch file.   (unless you want some contributors work coming in subdirectories...)
- etc.

These things become more important as you move past 4-5 contributors, so choose and implement at your discretion.

These type of practices are used often in distributed software development work, (except the software-team dev practices are: builds must be passing compilation with no errors before checking in a file, files have to follow naming standards, source code requires comments per function, etc.)

_Originally posted by @ddkaiser in https://github.com/TheGuyDanish/CM4_MATX/issues/15#issuecomment-729245879_

garlick, 2020-11-17:
Nicely stated.

I wanted to add that github workflows may be able to automate some of these checks on PR's, to the extent that they are scriptable.

TheGuyDanish, 2020-11-17:
https://github.com/marketplace/actions/kicad-exports It seems some people are already ahead of us on the automation front.

TheGuyDanish, 2020-12-28:
We've come quite a ways on this (implementing automated CI today, with the ability to enable DRC/ERC checks later when the board is in a better state to pass such tests), and I plan to write more contributing documentation soon.

I'll close this for now as we've made some good progress, and I have more TODOs lined up in line with it.

closed 2020-12-28 by TheGuyDanish

garlick, 2020-12-29:
Great work!