TheGuyDanish/CM4_MATX #19: CM4_MATX.kicad_pcb has hard-coded Windows-specific links to SamacSys 3dshapes

volkertb, 2020-11-17:
I noticed the following when I tried importing the KiCAD project in FreeCAD, using the KiCadStepUp addmon.

Apparently, in addition to the default library location in $KISYS3DMOD, this project has been configured to assume hard-coded locations of 3D images of some of the components in the default Windows installatoin folder of the SamacSys ibrary Loader.

If you type the following command:

```
cat CM4_MATX.kicad_pcb | grep -i samac
```
That results in this output:

```
    (model "C:\\SamacSys_PCB_Library\\KiCad\\SamacSys_Parts.3dshapes\\72309-8014BLF.stp"
    (model "C:\\SamacSys_PCB_Library\\KiCad\\SamacSys_Parts.3dshapes\\PI7C9X2G404SLBFDE.stp"
    (model "C:\\SamacSys_PCB_Library\\KiCad\\SamacSys_Parts.3dshapes\\USB2517-JZX.stp"
    (model "C:\\SamacSys_PCB_Library\\KiCad\\SamacSys_Parts.3dshapes\\39-28-1243.stp"
    (model "C:\\SamacSys_PCB_Library\\KiCad\\SamacSys_Parts.3dshapes\\SS-52100-002.stp"
    (model "C:\\SamacSys_PCB_Library\\KiCad\\SamacSys_Parts.3dshapes\\10104110-0001LF.stp"
    (model "C:\\SamacSys_PCB_Library\\KiCad\\SamacSys_Parts.3dshapes\\10018783-11200TLF.stp"
```

Obviously, this will only succeed in when you're running KiCAD in Windows, and even then only if the SamacSys ibrary was indeed installed in that folder.

Now ideally, we would want to include any 3dshapes that are not provided in the standard KiCAD llibrary, and refer to them through a closs-platform project-relative path. Unfortunately, there's the license of these 3D models to consider. The ones from SamacSys are apparently "free as in beer" (not sure what the exact terms are in the fine print), but that doesn't necessarily allow us to distribute them with these sources

For an optimal render that will look close to the actually envisioned end product, we need 3D models of the components that will go into the design.

Any ideas on how to tackle this?

TheGuyDanish, 2020-11-17:
This should be solvable with some more copying and pasting. I use SamacSys Library Loader for getting lots of models, schematics and footprints from Mouser. I just have to take note of this and remember to copy over the 3dshapes as well. For now, I think this depends a bit on #17, though.

TheGuyDanish, 2020-11-22:
Resolved by 8946d66.

closed 2020-11-22 by TheGuyDanish