TheGuyDanish/CM4_MATX #20: atx style power control

garlick, 2020-11-18:
Has there been any discussion about ATX style power control?  E.g. press the power button once and the OS shuts down, and then powers off.  Hold down the button to override the OS shutdown.

Here are a few references to some pi-related stuff that might be of use:

**soft power off**
This device tree overlay can be used to map a GPIO to key 116 (power key):
https://www.kernel.org/doc/Documentation/devicetree/bindings/input/gpio-keys.txt

pi specific discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=115394

You can then use a custom udev rule to map this to "power-switch" tag.  Systemd then gets ahold of the event and can initiate the shutdown, and finally take the action configured in logind.conf according to its HandlePowerKey setting, which defaults to "poweroff".

**turning off ps**
After the OS has shut down, you want to signal to the ATX power supply that it can turn off.  Here's a discussion about a pi-specific overlay that maps a GPIO pin such that it changes state once the OS has halted the CPU:
https://www.raspberrypi.org/forums/viewtopic.php?t=201483

Lots of ways to implement this, but I'll stop there - maybe this has already been thought about?  (Apologies if I missed it!)

mo-g, 2020-11-20:
Yes. We're (by which I mean @TheGuyDanish ) is already working on that, but hardware hasn't arrived yet to see how the power enable and reset pins behave under different conditions. I'm hoping to get a kit next week, with luck.

mo-g, 2020-11-20:
Just to summarise - I don't think we're looking to add a "shutdown key" in the sense of the ACPI "if you press this key, the system will exec `shutdown -h now`"- but instead, something to power the pi on after you shutdown from cli or normal graphical interface.

Also Run_PG as a hardware reset for a Pi that's locked up.

garlick, 2020-11-20:
Oh thanks for that reference.  To be honest I didn't follow it all, but I'm at least glad you're all over it.  IMHO, would be nice if one got more or less the same power button semantics as you get with a PC's power button, for principle of least surprise and all that.  I'm still a bit unclear if that's the goal or if you have something else in mind.

TheGuyDanish, 2020-11-20:
@mo-g Showing your age, buddy. ;) Jokes aside, my case still has a dedicated reset switch, they're just usually much smaller in size than the power one. So keeping it around is still a good idea, imo.

And yes, as was mentioned, I've got a CM4 and and IO Board incoming Soon(TM), as soon as they're in stock at Pimoroni. Once I've got it, I'll be looking into how GLOBAL_EN and RUN_PG behave and how they might be used for controlling power to the system. I'm still quite happy to take ideas and look at schematics if anyone has them.

geerlingguy, 2020-11-20:
@TheGuyDanish - Note that for the power situation, there's also on the IO board, J2 pins 13-14, which in the description state:

> Connect a push button to wake up the CM4 from low power mode. It can’t be used to shutdown the CM4.

Basically, when I do a `shutdown`, if I want to boot the Pi back up, I can just put a jumper on those two pins, when I pull it off, the Pi powers up. So it's not on button down, but on button up when it triggers the boot.

I haven't touched the J1 pins yet (in fact, I didn't even know where they were until I noticed you mentioning them!). There are no pins installed on the IO board, so I'll have to solder some up to see if they work as expected (I often have a need to hard-reset the Pi, so it would be nice to have for convenience's sake!).

mo-g, 2020-11-20:
> @TheGuyDanish - Note that for the power situation, there's also on the IO board, J2 pins 13-14, which in the description state:
> 
> > Connect a push button to wake up the CM4 from low power mode. It can’t be used to shutdown the CM4.

That's also linked to the GLOBAL_EN pin we've mentioned elsewhere, with a latch to RUN_PG to disable the pin when the system is powered up.

I hadn't actually realised this jumper existed, so when I described this exact concept in your issue, I was inventing it in parallel. :facepalm: I spent a lot of time thinking about how that would work, should have just read the IO board datasheet. :roll_eyes:

mo-g, 2020-11-20:
> @mo-g Showing your age, buddy. ;)

Aren't I just? I'm rapidly heading for my mid-thirties, it's scary...

mo-g, 2020-11-21:
> would be nice if one got more or less the same power button semantics as you get with a PC's power button, for principle of least surprise and all that. I'm still a bit unclear if that's the goal or if you have something else in mind.

Problem is - that's not something the Pi offers us. We can build a 1996-style ATX power button with nothing but a single logic gate - and the Pi IO board already does this. To build a 2003+ style ATX power button, we'd need to add our own logic - to count the amount of time the button is held down for, and send a shutdown key if the system is up; manually cut power if it's held down too long, power on if pressed while the system is off - $0.50 microcontroller and about 50 lines of C. No big deal - but as of right now, nothing on the board needs programming - solder, plug in CM4, go. Adding our own "Power Management Unit" would now mean IC's have to be programmed either before or after soldering, which is likely to be a much bigger cost than just the part itself.

I don't think we should rule it out; but we should be careful how much a force-off gesture is worth in terms of cost, as opposed to just pressing the reset button.

HaasJona, 2020-12-14:
It should be possible to build this simply in hardware without programming a microcontroller. Probably just needs a 555 or 556 timer ic and a few transistors or mosfets for the logic and cutting the power. The button could connect to the already present power-up gpio line, which could be polled by simple software to shut down the PI gracefully (if pressed shorter than 20 seconds).

TheGuyDanish, 2020-12-28:
I think to begin with, a more simple solution to go for, unless someone can contribute with a more involved schematic, would be to go for a simple PS_ON soft latch. If Global_EN is low, signalling the Pi is off, and the power button is pressed, latch PS_ON to GND. If Global_EN is high, signaling the Pi is on, and the power button is pressed, remove the latch, turning off the power supply again.

I discussed this with @funkyguy4000 on Discord (link in #29) and we agreed that while having full, ATX-style power supply control would be nice, managing the timers and power states in an electronic circuit isn't something we've had any solid idea for how to implement.

For now, I suggest we implement the latch circuit as mentioned above and use that moving towards revision 1. If, by the time we reach revision 1, someone has a more fully-fledged circuit to take care of the other concerns, I'd be happy to add it in. But in the spirit of getting this moving, I'm more willing to go with the other option.

TheGuyDanish, 2021-01-18:
Currently shopping for some components so I can test it out in a physical circuit, but I'm looking towards this as a potential option for the first board revision:
[image]

(Borrowed from here: http://www.mosaic-industries.com/embedded-systems/microcontroller-projects/electronic-circuits/push-button-switch-turn-on/latching-toggle-power-switch#press-on-hold-off-latching-circuits)

volkertb, 2021-01-20:
Is this ATX soft power circuit design from the Commander X16 project useful for the CM4_MATX project in any way?

https://youtu.be/Y-c3c3ia9Bo?t=297

bluerise, 2022-01-02:
I think the STM6601CM2DDM6F could be a good solution. It's a power-on push button controller designed for that specific use case: https://www.st.com/content/ccc/resource/technical/document/datasheet/1f/9e/3b/43/ec/04/44/71/CD00226957.pdf/files/CD00226957.pdf/jcr:content/translations/en.CD00226957.pdf

This design is used on the HoneyComb LX2K mini-ITX board (also ARM), and I think it would make sense here as well: https://developer.solid-run.com/download/clearfog-cx-and-honeycomb-schematics/ -> mitx-cex7-1.3.pdf

It provides an interrupt pin for 'short push', so that the OS can power down safely. On long-push it also provides that and after some time disables EN. The chip is powered by standby 5V, and controls the power-supply ON pin on the ATX header. The PS_HOLD pin can be used by the processor to force power-off. Seems like a nice setup.

On Discord concern was raised about the comment "The push-button controller is configured in a way that powers-on when ATX power is available.", that this should be made configurable, e.g. through a jumper. I wonder how it's implemented. Maybe C1/C2 pull down the button long enough when power comes in initially. Maybe adding a jumper before those would help make that configurable?

myself248, 2022-01-02:
There's obviously a fair bit of complexity here, so for rev 0, I'd like to put all this on a daughterboard, and just have a simple "short /PSON" bypass jumper for when you don't want to bother with all that. Then we can make boards carrying an STM660x, rp2040, atmega, esp, 555, who knows what else, and hammer out the desired functionality. 

(I initially suggested making this a standard footprint like micromod, and Danish suggested a Pi Pico footprint, but given that there are 5v and 12v signals in here, that's not so practical without support circuitry on the mobo, which kinda defeats the purpose. Maybe the signal conditioning wouldn't be onerous. Maybe a Pico-with-extension that the normal board wouldn't cover but a purposebuilt could....) 

Anyway, planning into that, I'm making a list of all the signals that might possibly be needed now or in the future for advanced functions. To start:

- GND, beacuse if I don't write it down I will forget it... 
- +5VSB from the ATX PSU
- +12VSB from the ATX12V PSU
- nPS_ON from the ATX/ATX12V PSU
- +5V from the ATX PSU
- +12V from that ATX/ATX12V PSU
- PWR_OK from the ATX/ATX12V PSU
- RUN_PG from the CM4 
- nRPIBOOT from the CM4
- PI_LED_nPWR from the CM4
- GLOBAL_EN from the CM4
- nEXTRST from the CM4
- WL_nDISABLE from the CM4
- UART pair for config? which?
- i2c/smbus pair for config? which?
- SPI triplet for config? which? 
- PWR_LED from the front panel header (is this two wires for bicolor antiparallel LEDs?) 
- HDD/ACT_LED from the front panel header
- PWR_BTN from the front panel header
- RST_BTN from the front panel header (note many cases lack this so don't depend on it) 
- possibly a chassis intrusion switch
- possibly a wireless-disable switch
- what else?