TheGuyDanish/CM4_MATX #22: audio and microphone jack

derwana, 2020-11-19:
One of the things the pi really lacks is good audio in and output. The Pi 4 model B got an audio jack, the IO Board none. But what about a microphone jack, 2.1 sound (or even 5.1). There are some HATs out there doing this by I2S, SPDIF or an HDMI Audio Grabber IC.

Some examples:
[SUPTRONICS ST6000](https://www.audiophonics.fr/en/dac-and-interfaces-for-raspberry-pi/suptronics-st6000-dac-es9023x4-hdmi-24bit-192khz-71ch-for-raspberry-pi-p-11849.html) or [X6000K-7.1CH Expansion Board](http://www.suptronics.com/miniPCkits/x6000-7.1ch.html) they seem to be the same,
[Amplifier HATs by HifiBerry](https://www.hifiberry.com/amps),
[HiFiBerry Digi (digital audio interfaces)](https://www.hifiberry.com/digis),
[Wolfson Audio Card](https://www.element14.com/community/community/raspberry-pi/raspberry-pi-accessories/wolfson_pi)

All these solutions are quite expensive but certainly worth considering for a Desktop PC.

TheGuyDanish, 2020-11-19:
While an interesting proposition, this is slightly felled by price and many of the ideas that were brought up in #3. While we could add things like audio connectors and all manner of fun ports. If we're exposing GPIO to the user (which is the current plan, having the GPIO header in place of a fourth PCI-e slot on the board), it would be up to the user if they'd like to add a board like this. Admittedly with some kind of adapter, most likely.

That said, as always, if others are interested in getting something like this on the board, I'd be happy to hear it.

TheGuyDanish, 2020-11-21:
I just had another thought on this. We have two USB2.0 expansion headers that allow for two ports each. I definitely see a world in which we could sacrifice one of these headers and leave open two ports for integration with whatever might be a neat feature. I know the phrase "USB sound card" probably doesn't inspire much confidence in audiophiles, but it _is_ a solution to adding some audio to the board, and it doesn't use up any precious GPIO pins.

At this point in time:
HD1 & HD2 are the USB ports used for the rear ports in the stacked Amphenol 72309-8014BLF connector
HD3 & HD4 belong to expansion header 1
HD5 & HD6 belong to expansion header 2
HD7 is the internal USB Type A that will be mounted vertically on the board. (Bel-Stewart SS-52100-002 connector)

My thought would be to discard expansion header 2 and leave HD5 and HD6 open for any kind of built-in functions. Anyone have input on this?

dogbreaf, 2020-11-22:
> I know the phrase "USB sound card" probably doesn't inspire much confidence in audiophiles

On the contrary, most audiophiles are going to be using an external DAC anyway since they avoid issues with electrical noise inside the case, and most of those are USB

How are the pi model B's providing analogue audio out? Because it seems like the easiest option (if it doesnt interfere with other planned features) would be to just copy that, and anyone who wants audio input could just use a dedicated USB soundcard or more professional USB audio interface

derwana, 2020-11-22:
[image]

It is taking the signals from `pwm1_mosi` and `pwm0_miso`, directs them through a bandpass (i think) on to the jack

TheGuyDanish, 2020-12-20:
I'm still more inclined towards a USB solution for this as, to my knowledge, the PWM1_MOSI and PWM0_MISO pins referred to above are actually GPIO pins, and that adding an audio interface like the one on the Pi4 could interfere with the use of those pins. (I'm not sure of this, so please correct me if I'm wrong). But for the above reason, I would personally prefer a USB-based solution that can be 'separate', if that makes sense.

funkyguy4000, 2020-12-20:
It may simplify the manufacturing and help bring cost down if we only offer the audio jack using the PWM pins mentioned above. That will still give a baseline audio option for those that need a simple output a d keep the USBs on the headers. If the user wants a better USB based experience, they can purchase a USB audio adapter. Chances are if they are using a raspberry pi for audio based projects stuff right now, they're using the built in or have an adapter they want already. We shouldn't choose their usb audio adapter for them

TheGuyDanish, 2020-12-20:
Good call on that one. In that case, I think re-implementing the kind of design used in the Pi would probably be suitable.

I'll look into this sometime soon, assuming no one else does it before me. (Trying to get PI7C power done during the next week.)

volkertb, 2021-01-26:
@derwana mentioned an HDMI Audio Grabber IC as one of the possible solutions. I kind of like that. It takes the audio signal already coming out of one of the HDMI ports and converts it to both digital S/PDIF and analog audio outputs.

Considering how devices such as the one shown below [go for a few bucks on Aliexpress](https://www.aliexpress.com/item/32780772075.html), my guess would be that the ICs they are based on are a dime a dozen and can probably be sourced very cheaply.

[image]

An integrated solution like this would be cheap, wouldn't require us to sacrifice any USB ports, would enable us to add both digital S/PDIF and analog sound output ports, without any of the Pi's native on-board sound playing capabilities posing any quality bottlenecks. The audio coming off the HDMI port(s) is digital after all, right? Also, HDMI audio doesn't have any of the latency problems that you might have with USB audio devices.

Perhaps similarly priced ICs also exist that perform the reverse action (take audio from an analog or S/PDIF source along with an HDMI video input and merge those into a combined HDMI signal that we could then feed into the TC358743 CSI2-to-HDMI bridge IC to provide audio input capabilities from microphone and line-in ports (both analog and S/PDIF) as well.

TheGuyDanish, 2021-02-10:
Moving this to revision 1.? rather than 1.0. I don't see this as being an important feature to the first iteration of the board, as I imagine the kinds of people that would want to get their hands on this in the first couple of runs aren't the type that would necessarily need 3.5mm audio jacks (or would be happy with a USB or PCI-e adapter). I'm still very much interested in getting this implemented, but I think there's a lot of ideas with no clear-cut solution at the moment.