TheGuyDanish/CM4_MATX #28: Power wiring for PI7C PCI-e switch

TheGuyDanish, 2020-11-27:
As I've mentioned a couple of times before, there are some parts of schematic and PCB design that I'm not quite confident with.

In the interest of moving this project forward, I'd like to request some assistance from anyone that has experience in designing schematics, specifically around powering ICs with different voltage. (1v, 3.3v, both analog and digital power)

I have requested, and received, a trove of design documents from Diodes Inc regarding this chip (That I am not publicly sharing due to concerns about disclosure). Including application notes, checklists, PCB design guides and a reference board design. I'd be happy to pass this onto anyone who would like to oversee getting the right power wired into this chip.

Alternatively, there is a chance I will be able to look at this around mid-December when I get back to Denmark and have a chance to chat with my father about it, as he's more qualified with these things.

TheGuyDanish, 2021-01-18:
As of fdb83ae, I dare say this is mostly done. I still would love if someone with a background or experience in power delivery could look through the PCI-e sheets and let me know in an issue if something here seems to be off or require adjustment. None the less. One step closer.

Durburz, 2022-01-05:
Although I don't have that much experience in power delivery circuits I would be willing to have a deeper look at this part if you could provide me the documents mentioned. (If all of this is still necessary)

TheGuyDanish, 2022-01-05:
I appreciate it, however @myself248 is currently working on potentially re-doing the PCIe switch section with a new chip in mind. He's in the Discord linked in the README, should you want to ask him yourself. Otherwise I can also point him to here.