TheGuyDanish/CM4_MATX #32: uatx dimensions

aep, 2020-12-17:
Hi, did you verify the uatx board yet? I might want to steal that :)

not sure what reference you used, but the holes dont line up with 
https://doc.xdevs.com/doc/_PC_HW/Form_factors/matxspe1.2.pdf page 12

TheGuyDanish, 2020-12-17:
The dimensions themselves should fit. 244x244mm. Mounting hole placement was done by @derwana in #21, using this as reference: https://cdn.instructables.com/ORIG/FS8/5ILB/GU59Z1AT/FS85ILBGU59Z1AT.pdf with (0,0) being the top left of the board.

I'm not sure if that might be incorrect. I'm not able to look at it until Sunday, as I'm currently getting ready for a flight back home for Christmas, but if someone could verify and potentially submit a pull request, I'd be happy to see it.

derwana, 2020-12-17:
In which way they don't line up? @aep watch out the projects board is rotated 90° to the left in comparison with the Layout Diagram you mentioned (used both of the files to do the positioning).

aep, 2020-12-17:
for example hole H2 should be 10mm from edge, not 5

[image]

derwana, 2020-12-17:
oh that's wierd – did i mix mm and inch?! I'll fix that tomorrow in a pull-request.

PR #33 (2020-12-20): fixing mounting hole placement
Fixes #32 mounting hole placement
closed 2020-12-20 by TheGuyDanish