TheGuyDanish/CM4_MATX #35: Atx12vo

ldrrp, 2020-12-20:
I dont mean to toss some logs into a fire but perhaps considering the pi is known for efficiency. We adapt the new trend of atx12vo power supplies. They tend to be cheaper and more efficient naturally since they only  dish out one voltage. Im not saying drop the normal atx only but maybe a variant board on the repo that could be kept in line with the main board with the exception of the power circuits.

If thats seen as not needed or wont do please close this or comment your thoughts so others know.

TheGuyDanish, 2020-12-20:
ATX12VO has crossed my mind when I was doing the initial legwork on this. Though since it hasn't proliferated as much yet among DIY builders. (Which I imagine will be the primary users of a board like this) I think sticking to regular ATX is the best bet for now.

That said, I think once we get to a point where we have a board in a higher revision, once everything is wired up, I'd be happy to look into the possibility of creating an ATX12VO branch. (Or if someone else wants to do this work now, they are welcome to present it)

Adding future revision label.

ldrrp, 2020-12-20:
This makes perfect sense. I can agree with branching at a higher revision as well.

TheGuyDanish, 2020-12-28:
Closing this in favor of #46 to clean up current issues.

closed 2020-12-28 by TheGuyDanish