TheGuyDanish/CM4_MATX #38: Errors in RJMG2012119A0NR.stp

derwana, 2020-12-24:
The STEP file `RJMG2012119A0NR.stp` is kind of odd. Footprint and device do not match (through holes do not align with pins).

[image]

TheGuyDanish, 2020-12-24:
Looking into this, it appears to be an oversight of mine, more than anything. Rotating the device to fit on the bottom, rather than on top of the board, the 3D model aligns perfectly with the pins, so perhaps, this specific SKU was meant to be mounted to the bottom of a board.

Regardless, I'm still not sure I'm a fan of this port. RJ45 jacks have in general been the bane of my existence for the last little while. The previous one used in the design, the one by Bel-Stewart was 100 Mbit/s, which is why I replaced it. This one matched the description of having an on-board transformer, being gigabit and not having extra cost from being a PoE part.

I'll have a look at other potential parts available on Mouser and update the design with one that fits the board a little better, hopefully today or tomorrow.

(Merry Christmas to those celebrating, by the way.)

TheGuyDanish, 2020-12-24:
Think I got a proper candidate.

ARJM11C7-502-AB-ER2-T

https://abracon.com/Magnetics/ARJM11.pdf
https://www.mouser.co.uk/ProductDetail/ABRACON/ARJM11C7-502-AB-ER2-T/?qs=%2Fha2pyFaduic6SCsIt%2FVwE5Crt%2FwJyMPuBAgLXc7jMk%252BDddfCGi7wNa%2FpJ2Iv1iD

TheGuyDanish, 2020-12-24:
Only slight gripe with this was that I couldn't find a detailed 3D model anywhere on Abracon's website. I'm sure one exists somewhere on the Internet, but I'm not fussed enough to find one at the moment.

commit c230aee3 (2020-12-24): Replaced Ethernet connector and removed unnecessary files

Fixes #38
closed 2020-12-24 by TheGuyDanish