TheGuyDanish/CM4_MATX #42: voltage of GPIO_VREF not defined

derwana, 2020-12-27:
GPIO_VREF is used on sheets (1,) 2 and 6 but never defined.

Screenshots from CM4_IO:
[image]

[image]

TheGuyDanish, 2020-12-27:
J1 on CM4_HighSpeed
[image]

I do want to potentially make this a solder jumper to 3.3v by default.

TheGuyDanish, 2020-12-27:
Actually, disregard. Just read that schematic again. Yes, we do need to establish VREF voltage. Perhaps via the aforementioned jumper

closed 2020-12-28 by TheGuyDanish