TheGuyDanish/CM4_MATX #49: Implement Real Time Clock

TheGuyDanish, 2020-12-28:
As mentioned in #1, we can quite easily implement an RTC into our design (which I believe would be a genuinely nice and useful feature). I'm breaking it out into a new issue to increase visibility.

> On-board RTC is present in the CM4IO board and could easily be lifted if we want to integrate it.
> [image]
> [image]
_Originally posted by @TheGuyDanish in https://github.com/TheGuyDanish/CM4_MATX/issues/1#issuecomment-727240965_

TheGuyDanish, 2020-12-29:
Looking further into this, I think using the NXP [PCF85263A ](https://www.mouser.dk/datasheet/2/302/PCF85263A-1127483.pdf) instead would be a better call. It has built-in circuitry for switching to battery power, meaning we can power the RTC with 3.3vSB when the power supply is powered and a coin cell battery when it isn't.

derwana, 2020-12-30:
I'll take a look into it an do a schematic etc. in the new year.

TheGuyDanish, 2021-01-06:
Closed by #51

closed 2021-01-06 by TheGuyDanish