TheGuyDanish/CM4_MATX #52: No assigned footprints

derwana, 2021-01-05:
Output from 'update PCB from Schematic':

```
Error: Cannot add C65 (no footprint assigned).
Error: Cannot add C45 (no footprint assigned).
Error: Cannot add C35 (no footprint assigned).
Error: Cannot add C36 (no footprint assigned).
Error: Cannot add C44 (no footprint assigned).
Error: Cannot add C37 (no footprint assigned).
Error: Cannot add C38 (no footprint assigned).
Error: Cannot add C43 (no footprint assigned).
Error: Cannot add C40 (no footprint assigned).
Error: Cannot add C42 (no footprint assigned).
Error: Cannot add C41 (no footprint assigned).
Error: Cannot add C50 (no footprint assigned).
Error: Cannot add C33 (no footprint assigned).
Error: Cannot add R3 (no footprint assigned).
Error: Cannot add C46 (no footprint assigned).
Error: Cannot add J22 (no footprint assigned).
Error: Cannot add C32 (no footprint assigned).
Error: Cannot add J23 (no footprint assigned).
Error: Cannot add C31 (no footprint assigned).
Error: Cannot add C47 (no footprint assigned).
Error: Cannot add C30 (no footprint assigned).
Error: Cannot add C29 (no footprint assigned).
Error: Cannot add C48 (no footprint assigned).
Error: Cannot add R1 (no footprint assigned).
Error: Cannot add R28 (no footprint assigned).
Error: Cannot add C27 (no footprint assigned).
Error: Cannot add R4 (no footprint assigned).
Error: Cannot add C56 (no footprint assigned).
Error: Cannot add C26 (no footprint assigned).
Error: Cannot add R6 (no footprint assigned).
Error: Cannot add C49 (no footprint assigned).
Error: Cannot add C25 (no footprint assigned).
Error: Cannot add R8 (no footprint assigned).
Error: Cannot add R9 (no footprint assigned).
Error: Cannot add C24 (no footprint assigned).
Error: Cannot add R10 (no footprint assigned).
Error: Cannot add C8 (no footprint assigned).
Error: Cannot add Y1 (no footprint assigned).
Error: Cannot add C23 (no footprint assigned).
Error: Cannot add R12 (no footprint assigned).
Error: Cannot add C68 (no footprint assigned).
Error: Cannot add R13 (no footprint assigned).
Error: Cannot add C22 (no footprint assigned).
Error: Cannot add R14 (no footprint assigned).
Error: Cannot add R15 (no footprint assigned).
Error: Cannot add C21 (no footprint assigned).
Error: Cannot add C67 (no footprint assigned).
Error: Cannot add R17 (no footprint assigned).
Error: Cannot add C20 (no footprint assigned).
Error: Cannot add C54 (no footprint assigned).
Error: Cannot add C57 (no footprint assigned).
Error: Cannot add C7 (no footprint assigned).
Error: Cannot add R19 (no footprint assigned).
Error: Cannot add C19 (no footprint assigned).
Error: Cannot add R20 (no footprint assigned).
Error: Cannot add C66 (no footprint assigned).
Error: Cannot add R21 (no footprint assigned).
Error: Cannot add C18 (no footprint assigned).
Error: Cannot add R22 (no footprint assigned).
Error: Cannot add R23 (no footprint assigned).
Error: Cannot add C53 (no footprint assigned).
Error: Cannot add R24 (no footprint assigned).
Error: Cannot add R11 (no footprint assigned).
Error: Cannot add R25 (no footprint assigned).
Error: Cannot add C16 (no footprint assigned).
Error: Cannot add R26 (no footprint assigned).
Error: Cannot add C17 (no footprint assigned).
Error: Cannot add C52 (no footprint assigned).
Error: Cannot add C14 (no footprint assigned).
Error: Cannot add R29 (no footprint assigned).
Error: Cannot add R27 (no footprint assigned).
Error: Cannot add C64 (no footprint assigned).
Error: Cannot add C15 (no footprint assigned).
Error: Cannot add C28 (no footprint assigned).
Error: Cannot add C51 (no footprint assigned).
Error: Cannot add R2 (no footprint assigned).
Error: Cannot add J21 (no footprint assigned).
Error: Cannot add C63 (no footprint assigned).
Error: Cannot add C12 (no footprint assigned).
Error: Cannot add R16 (no footprint assigned).
Error: Cannot add C11 (no footprint assigned).
Error: Cannot add C62 (no footprint assigned).
Error: Cannot add C10 (no footprint assigned).
Error: Cannot add U2 (no footprint assigned).
Error: Cannot add C9 (no footprint assigned).
Error: Cannot add C61 (no footprint assigned).
Error: Cannot add C34 (no footprint assigned).
Error: Cannot add R18 (no footprint assigned).
Error: Cannot add C13 (no footprint assigned).
Error: Cannot add C60 (no footprint assigned).
Error: Cannot add C6 (no footprint assigned).
Error: Cannot add C5 (no footprint assigned).
Error: Cannot add C59 (no footprint assigned).
Error: Cannot add C4 (no footprint assigned).
Error: Cannot add C55 (no footprint assigned).
Error: Cannot add C3 (no footprint assigned).
Error: Cannot add C58 (no footprint assigned).
Error: Cannot add C2 (no footprint assigned).
Error: Cannot add C1 (no footprint assigned).
```

TheGuyDanish, 2021-01-05:
I'm aware of this. For the most part, this is down to sizes not having been decided yet for most components. As a result of which, there's no footprint assigned. We could go through this and sort it fairly soon, but given that most components have been randomly numbered, the thought of renumbering components manually have occurred to me, but I don't know if it's worth the effort for an early board.

TheGuyDanish, 2021-02-10:
Closing this in favor of task #59 for now.

closed 2021-02-10 by TheGuyDanish