TheGuyDanish/CM4_MATX #53: heat sink for CM4

garlick, 2021-01-06:
Has heat sink mounting been discussed for this board?

I've looked around and not found much offered specifically for the CM4, but I have made contact with  [Alpha Nova Tech](alphanovatech.com) and requested some help fitting one of their products to a CM4 for another project.  Right now it's looking like M2.5 shoulder bolts and press-fit PEM style threaded inserts, with an off the shelf heat sink, custom drilled for the CM4 will be the most economical solution.

Would appreciate any insight others have here, and of course will share what I learn if that's helpful.

geerlingguy, 2021-01-06:
With active cooling (e.g. a fan anywhere in the case), I haven't had any trouble keeping my CM4 from throttling. For other Pis, I typically just use a little heat spreader like the ones that come in most cooling kits, that's about the dimension of the chip itself, about 3-5mm high.

garlick, 2021-01-06:
Good to know!  Maybe it won't be a problem here.

TheGuyDanish, 2021-01-07:
My initial thought with this was to just use the mounting holes that are already on the CM4. The Turing Pi 2 already illustrates a heatsink design that uses those mounting holes, and I imagine solutions for them (perhaps with a plastic backplate and screws on top, like a regular CPU cooler) will appear in time.

[image]

Alternatively, if someone just wants to put a single small heat spreader/heatsink on like Jeff does, I think that should be fine as well. My main concern with regards to cooling is just getting airflow through the case.

garlick, 2021-01-07:
Yeah this Turing Pi pic is what I sent the Alpha Nova Tech guy as a starting point.  Apparently a normal PCB thickness (1.6mm) is not sufficient thread depth for those shoulder screws, hence the recommended inserts., but guess a backplate could do the trick too if there's enough room under the board.

TheGuyDanish, 2021-01-07:
>if there's enough room under the board.

It would need a bit of engineering to make a custom solution, but most cases sold these days have some room (or even a full cut-out) behind the CPU specifically for easy backplate mounting. The typical Noctua one is not much thicker than 3-4mm, putting the mounting pegs up through the holes for you to screw the cooler down onto.

This shows what I mean fairly well: https://youtu.be/3qq9syfjxc8?t=137

mo-g, 2021-01-14:
I guess the only thing that @TheGuyDanish would need to make sure here is that there were no components standing proud under the board where the SOM mounts, and that some large diameter holes for a backplate's studs are available.

After that, assuming the PCB thickness of this board matches other people's PCB thickness, anyone elses' heatsink solution for the CM4 should be a drop-in.

derwana, 2021-02-10:
[Elecrow](https://www.elecrow.com/heatsink-for-raspberry-pi-compute-module-4-cm4.html) and [AliExpress](https://de.aliexpress.com/item/1005001933362417.html) now offer some heatsink(case)s for the CM4.

TheGuyDanish, 2021-06-02:
Update to this, as expected, coolers that use the default CM4 mounting holes are starting to show up around the internet, with a couple of examples like these:
https://www.waveshare.com/cm4-fan-3007.htm
https://www.waveshare.com/product/cm4-heatsink.htm

Seeing as these are pretty naturally fit for the topic, I don't believe there's a reason to keep this issue open for now.

closed 2021-06-02 by TheGuyDanish