TheGuyDanish/CM4_MATX #56: ISA bus instead of extra pci?

tannewt, 2021-01-10:
What if you did one pci slot and then ISA? That would make this more of a retro hybrid kind of like the commander x16. You could use old ISA cards then too.

TheGuyDanish, 2021-01-10:
Hi Scott, and thanks for the suggestion!

While I think an ISA slot could be an interesting addition, I just don't know if it would be used enough to make it a worthwhile addition compared to the extra cost, in terms of components and the like. I see cards already exist to do USB<->ISA, which might be a better bet, in my opinion. The idea behind using the PCI-e switch to get more ports is two-fold. On one hand, it is to fill out the expansion slots in an mATX chassis, but on the other, it is also to help people who would like to try and debug or develop drivers for multiple cards at once for compatibility.

Perhaps a future revision of the board could implement a design like [Manawyrm/ISASTM](https://github.com/Manawyrm/ISASTM/) if we have spare space on the board at such a time, but other than that, I see the potential for us adding ISA capability to be quite limited.

TheGuyDanish, 2021-02-10:
Closing this for now to clean up inactive issues. Feel free to continue this topic in [Discussions](https://github.com/TheGuyDanish/CM4_MATX/discussions).

closed 2021-02-10 by TheGuyDanish