TheGuyDanish/CM4_MATX #60: How hard is/was this project to implement?

bhansconnect, 2021-02-20:
This isn't really an issue, but I was hoping to get some input because I am interested in doing something similar as a hobby project.

Just in general, how hard is it to build a board around the CM4? I would assume that if you stay within the default functionality it is relatively simple, but adding things like a PSU, DAC, or chips on the PCIE bus greatly increase the complexity.

What would you say are the hardest parts of designing this board? Any specific major issues you faced?
Have you gotten to the point of manufacturing any tests PCBs?  Have you been picking components specifically thinking about manufacturing?
Is there anyway for you to test things before manufacturing a PCB? I assume simulation of this with many black box chips would be extremely hard. Is the only option lots of schematic review to hopefully catch bugs?

Sorry for the barrage of questions, I am just hoping to learn more about the challenges of this sort of process.
Thanks for your time.

TheGuyDanish, 2021-02-20:
TL;DR: 
- While it requires some serious thought, practice and math. If a more skilled electrical engineer were to lead this project, they would likely have finished a first prototype and be bordering on a shipable product by now.
- The project would greatly benefit from an easy way for a home user to manufacture and assemble a board that uses these incredibly tiny and difficult to solder components, or access to someone with the necessary facilities to do this.
- To the above point, getting a board manufactured is the most inexpensive part. Home assembly could be done given a stencil and a reflow oven (a small model of which likely wouldn't be of great difficulty to acquire). This is something I might look into, but as I'm currently planning to move out of London, it'll be pending my arrival in my new home.

Let me know if any other questions arise from this and I'll be happy to answer.

---

I'll preface my answer to this with the following: I am by no means a qualified electrical engineer and have only a small bit of experience with schematic design, and less with actual PCB routing. My experience is purely self-taught, though I have a grasp of basic electrical theories (Ohm's law, etc), mostly with regards to mains voltage work, logic gates and some Arduino-style microcontroller programming. With that said, I believe in the sort of maker/hacker ethos that is "No matter who you are or what your experience level is. Somewhere, someone doing what you're doing in a professional capacity is bodging shit the exact same way you are."

Onto the actual answers, though.

> Just in general, how hard is it to build a board around the CM4?

My honest opinion on working with the CM4 so far (without having had a board I've designed in my hands yet) is that it seems easy on the surface, but I'm waiting for any small 'gotchas' to show themselves. The CM4 has excellent documentation and the IO board is an amazing resource for drawing examples from. A lot of schematic work here has been largely inspired by, if not lifted from, the CM4 IO Board's open source design files.

I think the difficulty does increase when you try to add in things that have a certain standard or perceived idea of how they should work. The example of ATX-style power control comes to mind. Everyone knows how a power button on a PC, ideally, should work. However implementing that functionality without the use of a microcontroller is a challenge, especially if you're not 100% up on your ins and outs of 555 timers. In newer motherboards, the signal from the power button "disappears" into the chipset, which is a black box for all we know.

I genuinely don't think the addition of the PI7C PCI-e switch chip has made the design more complicated. The chip is well-documented and within an hour of sending them an email, the manufacturer of the chip (Diodes Inc) sent me a full design kit. It contains a bunch of useful information including a reference board design using the chip, as well as several guides on PCB routing for PCI-e that is useful for more than just that chip.

> What would you say are the hardest parts of designing this board? Any specific major issues you faced?

By far, my biggest issues with this board are my insecurities around prototyping it. As someone who doesn't have a great deal of practical experience designing boards, I would love to test the schematics put together for this project. However, because of price (and part complexity), most components are surface-mount devices that are pain in the neck to solder by hand. Not saying it is impossible, not at all, but a challenge.

In theory, it would be possible to grab a schematic (let's say the PCI-e portion, or the USB hub) and send it off to a manufacturing house (PCBWay, for example) and have them assemble a prototype board specifically for that portion. But at that point, we would need to design a new board with just the parts we want to test, send it off to have it made and assembled, then shipped to someone who can test it. I actually do think this would be beneficial, as I am only able to dedicate a small amount of time to this project a week, so it wouldn't impact the overall timescale.

I'll continue to answer this question in the next section as well.

> Have you gotten to the point of manufacturing any tests PCBs? Have you been picking components specifically thinking about manufacturing?

Not yet. I have, by myself, created some test boards for a few individual features, but there is no "CM4_MATX Prototype Board" in existence at the moment. The main reason for this is because of the number of parts on the board. If we were to assemble this board as it stands right now, I estimate we would be looking at $50-80 in just parts. On top of that we would pay a, comparatively, small charge to have the board made, and another charge to have it assembled. This is why I'm hesitant to make a prototype board before I have a state I'd like to "lock", in effect, and show to an engineer that can tell me any obvious concerns. I don't want to sink between $200-400 into making 2-5 boards that will have a defect that would have been spotted by a more senior engineer.

With regards to components. Most of the named components in use on this board (fan controller, PCI-e switch, USB hub, etc) were picked mostly to match a use case. We want a feature-rich board with cost coming in second. This doesn't mean I want the most expensive parts I can get, but any part with a good reference design or [... cut ...] guess based on that information and prior knowledge about if it would work or not.

Aside from getting hands-on with the board itself, there are very few ways to actually catch a complicated issue before it shows itself on real hardware. This is also the reason why the first run of boards for this project will be very, very limited. Even if people are passionate about this as potential platform, I don't want people to sink money into buying a board that might have an obviously overlooked defect.

TheGuyDanish, 2021-02-20:
As you mention being interested in starting hobby projects yourself, I'd encourage you to walk in the footsteps of others first. One of my first fully-assembled PCBs is a USB to RS232 adapter using an FTDI FT232R chip. It has been done more times than we can count, but it is a useful exercise in both schematic design and PCB layout (especially if using a level shifter).

If you're doing projects that can use mostly through-hole components, but **needs** a few really small SMD ICs, you could use a manufacturing house to get a board with only the SMD parts populated, then do the rest yourself. If your SMD part doesn't have hundreds of pins, you could even break it out into breadboard-compatible 2.54mm (100 mil/thou) pins.

bhansconnect, 2021-02-21:
Wow, thanks for the super detailed response.

I am definitely not a qualified electrical engineer either. I am a software engineer who just happens to really enjoy EE stuff as a hobby. I have done some minor PCB design before, a relay adder, but otherwise have not done much with PCBs. That being said, I definitely have a lot of background in practical EE and have done a lot of random arduino style projects. I have also done a decent bit of FPGA work. I feel like I technically have the know how to design a basic board around the CM4, but their are a lot of easy to make mistakes that would lead to parts of the board not working an a lot of pain with debugging. This is one of the reason why schematic and PCB design has always been so weird to me. Though it is possible to test and to simulate, a lot of the chips are complex and essentially black boxes in a way that make simulation really hard. From simply swapping two pins to a component with a misread spec to running wires in a poor way, it is quite easy to make broken hardware.

What I actually want to make would essentially be a basic laptop board with the CM4 dealing with a significant part of the complexity. The goal would be to have CM4 + low profile mechanical keyboard +  port to directly connect lcd display as opposed to hdmi + battery management section(I think this is where most of the dragons would be) +  a few other random ports, most of which are built into the pi already. I know that this would be quite a huge project, but luckily it is possible to make individual parts. Also, there are open source mechanical keyboards already to build off of.

I feel like this project may fall into the doable, but insane to jump into category. Also, a ton of it can be done in software before I actually have to spend any money, which is nice. That being said, I am also afraid of sinking between $200-400 into making 2-5 boards and then them being broken. Which is why I may start by trying to make single sections of the board. Like oh here is a board that is just a cm4 and whatever it takes to connect to an lcd display. Then I would flash the cm4 with the io board and see if it can display over the lcd connector. Same with keyboard, etc. The weird part is that it will probably cost more overall to develop all the smaller boards assuming I could properly verify the big board before making the first version.

I feel like the main issues I would run into would fall into 3 categories: proper component selection, misreading/understand a spec, stupid mistakes. The problem is that if not caught, all of them are expensive and potentially hard to debug issues.

bhansconnect, 2021-02-21:
As a side note, I recently became aware of the pikeeb project, which is definitely similar in goal to what I want to do, but much larger and bulkier at a minimum. I guess I would be designing a lot of similar hardware though.

TheGuyDanish, 2021-02-21:
I think if you're looking to seriously get into board design, forking out the $200 or so for an infrared reflow oven on ebay is a good investment. Alternatively, you can also get a hot air gun, which should be suitable for simple reflow tasks, which is probably what you'd be doing with smaller/prototype boards anyway. If you can keep a board below a certain size, you could easily get five boards and a stencil for < $30-50 from JLCPCB. Of course, there will be some manual assembly, but as hobbyists, I think spending time with our projects is just part of the fun of it.

From a first reading, your board idea does make sense, and I do definitely think it is quite doable. Seeing as the Pi has onboard ethernet and USB (which you'd likely need to route to a USB hub) the most you'd probably have to think about is the LCD connection and the charging circuit, definitely.

Given the HDMI is exposed as just signals, you could repurpose another connector (say something similar to the flat connector used by the DSI header) to carry the signal to compatible display. You could also find a DSI-compatible display, but I read that most modern LCDs will require more than two pairs of data lines (which is all we've got per DSI header on the CM4). I'm almost certain there exists a good battery solution already using Li-Ion batteries. The theory is all out there, it just needs to be assembled.

All in all, it sounds very doable, and I think one of the best ways to see it through to reality is just to start tinkering. Find the bits and pieces you want, make an outline of what ports you'd like to have on-board and what ICs you'd need to implement them and go from there. Get the ICs, get some reference designs, make a schematic and make a demo PCB that uses a CM4 and that schematic information.

If you're using Discord, there's a link in the Welcome post in the discussions forum with a server I use to chat about this project. Feel free to join and shoot me a DM if you'd like to go over ideas, I'd be happy to contribute my help whenever possible.

bhansconnect, 2021-02-21:
I'm not sure I know which Welcome post you are talking about. Could you share a link?

TheGuyDanish, 2021-02-21:
[This one](https://github.com/TheGuyDanish/CM4_MATX/discussions/29)

> [image]

TheGuyDanish, 2021-03-01:
Closing this off for now as I think the question has been answered.

closed 2021-03-01 by TheGuyDanish