TheGuyDanish/CM4_MATX #71: ,

erkinalp, 2022-07-24:


TheGuyDanish, 2022-07-26:
So this is part of a larger bit of work being required, but addressing each of these in a TL;DR:

- SATA: Definitely something I'd like to do if we have the spare IO.[1]
- M.2: Also something I'd like to do.[1]
- Front USB 2.0 header: This is already implemented as J9 on the bottom corner of the board.
- Onboard audo device: Myself (or someone else) will need to do research into what will go into providing a 3.5mm jack out/in. last I recall, this required using an I2S line that's also by default used by PWM. Perhaps we could steal a USB port for it.

Expanding on [1]:
We've talked in the Discord (me and 'Myself' on Discord, confusing name, I know.) about what exact PCI-e switch chip to use. Initially, we thought the PI7C9X2G404SL was a good choice as it allowed us 1 upstream port and 3 downstreams. This only has a minor caveat in that we don't have enough PCI-e I/O to provide all four ports that are the theoretical max of the mATX platform.

In order to get around this, we talked about the PI7C9X2G606PR. This would allow us a total of six ports. As in, 1 upstream and 5 downstream. With this, we could theoretically provide:
- 4 PCI-e x1 slots + 1 M.2 (over PCI-e)
- 4 PCI-e x1 slots + SATA (via PCI-e bridge like the Marvell 88SE9xxx-series)
- 3 PCI-e x1 slots + 1 M.2 (over PCI-e) + SATA (via PCI-e bridge like the Marvell 88SE9xxx-series)

For feature richness, the last option would be preferred, however my main concern with adding SATA is the availability of Marvell's 88SE9xxx controllers. They're ideal for this task but have two huge points against them:
- They're in excessive demand as everyone is using them, and thus high price.
- While the datasheet is publicly available, it is not stocked by major retailers.

In chosing to use the 88SE9xxx series, we'd have to decide if we're going to prioritize making the board easy to get parts for, by only using parts from distributors like DigiKey, Mouser, etc. Or if we'd let users who want to assemble their own board try chips from random sellers on AliExpress/ebay/Alibaba or part scalpers?

On a similar note, I've been chatting with deinehardware who is also in the Discord about using an ASM1184e. However the drawbacks of this chip are the same as the Marvell ones.

I am starting a new job in 12 hours and so, don't really have much time to put towards this, but I'd like to hear opinions on this. Once things have settled down for me, I might look to tear off the page and start fresh on some of the schematics and goals for this project.

volkertb, 2022-07-26:
First of all, thanks for the update!

Could I make a case for option 1 (4 PCI-e x1 slots + 1 M.2 (over PCI-e)), though?

NVMe SSDs are not only faster than SATA SSDs, but also cheaper these days. So that would still make NVMe drives preferable over SATA ones, even if currently available generations of the Compute Module aren't yet able to make optimum use of them speed-wise.

That would leave optical drives as the only remaining compelling reason to have SATA connectors integrated on-board, but less and less people are even bothering to install optical drivers in they desktop PCs these days.

People who really need SATA could always add it by installing an PCIe card. Those who don't need SATA would then at least still have the option to use that fourth PCIe slot for something else. Plus that would make the motherboard cheaper to manufacture, so win-win.

So TL;DR: going for 4 PCIe slot + 1 M.2 (NVMe) slot and not integrating SATA would make the board both cheaper and more flexible in terms of expandability.